libremesh/chef #7: check the default settings for ppp and usb

FreifunkUFO, 2017-10-13:
when trying jefe f.i. for "Ubiquity Loco" there are packages for ppp and usb.
where does it come from? these devices dont habe USB

aparcar, 2017-10-14:
thank you I'll check that

aparcar, 2017-10-16:
using `make info`
```
Default Packages: [...] ppp ppp-mod-pppoe [...]

[...]

ubnt-loco-m-xw:
    Ubiquiti Loco XW
    Packages: kmod-usb-core kmod-usb2
```

The build-server only parses the output of the imagebuilders, is it possible the imagebuilder is setup bad? @dangowrt

Removing the packages from the "Edit packages" section results in an image without the packages :)

aparcar, 2017-10-16:
@FreifunkUFO this seems to be a LEDE bug, please report it here: https://bugs.lede-project.org/

closed 2017-10-16 by aparcar

dangowrt, 2017-10-16:
https://github.com/lede-project/source/blob/master/target/linux/ar71xx/image/ubnt.mk#L40

FreifunkUFO, 2017-10-16:
seems to be configured by @blogic when reordering 
https://github.com/lede-project/source/commit/636089ead6864364798ac988299429d171051c0e#diff-8fd5a4a2c103925f08c02265642dc96e