libremesh/chef #8: lime: wpad-mini vs. wpad-mesh

FreifunkUFO, 2017-10-16:
when using the 11s flavour, an other wpad* is needed. the needed package is added, but wpad-mini isnt needed anymore

[image]

FreifunkUFO, 2017-10-16:
the extra package: https://github.com/libremesh/network-profiles/blob/master/libremesh/encrypt-11s/PACKAGES

the imagebuilder will get from jefe:

Packages: base-files busybox dropbear firewall fstools ip6tables iptables kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ledtrig-usbport kmod-usb2 libc libgcc lime-full logd mtd netifd odhcp6c odhcpd opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch wpad-mini wpad-mesh base-files busybox dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ledtrig-usbport kmod-usb2 libc libgcc logd mtd netifd odhcp6c odhcpd opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch wpad-mini

see https://betaupdate.libremesh.org/static/lime/17.06/ar71xx/generic/tl-wr1043nd-v1/lime-17.06-4961fcbcd0d317f-libremesh-encrypt-11s-ar71xx-generic-tl-wr1043nd-v1-sysupgrade.bin.log

aparcar, 2017-10-16:
does the lime-sdk/cooker behave the same way? it's possible to remove `wpad-mini` via *Edit packages*, tho it's not done automatically

ilario, 2017-10-16:
related discussion on libremesh/lime-packages#208

FreifunkUFO, 2017-10-17:
problem on jefe.tk: the additional packages are not shown to the user.

aparcar, 2017-10-17:
yes, currently the packages are processed on the server side. I don't like that solution and will see if I can do something about it. however, a clean solution would be to auto remove the wpad-mini and this isn't supported for now with the PACKAGES concept

aparcar, 2017-11-02:
this issue is out of scope for this project :(

closed 2017-11-02 by aparcar

FreifunkUFO, 2017-11-03:
so what does it mean? problem is still existing. or do you want to delete libremesh as a supported "distribution"?

aparcar, 2017-11-03:
@FreifunkUFO problem still exists but it's not a problem of _chef_. Using the official _lime-sdk_ results in an image with both `wpad-mesh` and `wpad-mini` installed. The problem are thedependencies/flavors that should be solved within LibreMesh and not individually by _chef_.

ilario, 2017-11-07:
Can anyone open an issue on this on the correct repository? (lime-sdk?)

aparcar, 2017-11-08:
@FreifunkUFO would you mind?

FreifunkUFO, 2017-12-28:
@aparcar why is "this issue .. out of scope for this project :("?
otherwise please reopen ticket.