libremesh/chef #13: a new name for this project

FreifunkUFO, 2017-10-30:
the name "imagebuilder" is already existing for Openwrt/LEDE. Using the same name would be confusing. (see discussion on lime-dev mailinglist).
also the URLs should change, when a new name was found.

FreifunkUFO, 2017-10-30:
its a blocker for https://github.com/libremesh/lime-sdk/issues/32

aparcar, 2017-10-31:
I like "imagebuilder online". It's generic enough to fit for any distribution.

aparcar, 2017-10-31:
@ilario

ilario, 2017-10-31:
Uh!
Let's think as a possible subdomain for LibreMesh/LEDE/OpenWrt websites which could host the software.
From that we can fiddle around for finding the software name.
Brainstorming:

On the concept of making/building

1. builder.something.org
1. onlinebuilder.something.org
1. build.something.org
1. imagebuilder.something.org
1. maker.something.org
1. make.something.org
1. generator.something.org
1. onlinegenerator.something.org
1. cast.something.org
1. caster.something.org
1. crucible.something.org
1. foundry.something.org
1. create.something.org
1. creator.something.org
1. onlinecreator.something.org
1. cook.something.org

On the concept of customizing/tailoring

17. customize.something.org
1. custom.something.org
1. tailor.something.org
1. doctor.something.org
1. tweak.something.org
1. reshape.something.org
1. compose.something.org
1. craft.something.org
1. crafts.something.org

On the concept of a service

26. ondemand.something.org
1. prepare.something.org
1. order.something.org
1. justcook.something.org

FreifunkUFO, 2017-10-31:
or something with "attended" (according to neighbor-project "attended-sysupgrade"
30. attended.something.org
31. attendedfirmware.something.org

FreifunkUFO, 2017-10-31:
@aparcar "online" is not nice, f.i. because project also could stay in an offline-mesh to serve new images. the main difference to the nowadays existing "imagebuilder" is, that this project is a **webinterface** (to use the imagebuilder simplier than command-line-wise).
32. web-imagebuilder.something.org
is jefe also running on openwrt/lede and with luci-interface? so we could call it luci-imagebuilder :-o

aparcar, 2017-10-31:
Tempting idea to build a luci app ;) 
I like tailor.something. upgrade.libremesh.org for the server backend?

ilario, 2017-10-31:
More ideas, on food (the concept here is that you can assemble your images as you prefer, like a pizza):

33. pizza.something.org
1. pizzadelivery.something.org (these two were from @dangowrt on the lime-dev list)
1. freepizza.something.org
1. kebab.something.org
1. icecreams.something.org

Some on the cocktails naming of the OpenWrt releases:

38. brewery.something.org (sounds quite similar to the famous Homebrew software)
1. shaker.something.org
1. muddler.something.org
1. glassware.something.org
1. mixer.something.org (not really cocktails)
1. soda.something.org
1. distillery.something.org
1. bartender.something.org

@aparcar proposed 46. easy.something.org 'cause it will be an easy way for getting the needed image, even easier than the current downloads.something.org
on the same concept:

47. wizard.something.org
1. facile.something.org
1. simple.something.org

Or on the concept of downloading:

50. fetch.something.org
1. get.something.org
1. pick.something.org
1. downloadwizard.something.org

FreifunkUFO, 2017-10-31:
oh now i see, this is just the web-client. server is atm the "server" part of attended-sysupgrade.
54. attended-sysupgrade-tailor...

nicopace, 2017-10-31:
and why not chef? i really believe that chef is a nice name, and replacing the old with the new is a sane process...

aparcar, 2017-11-01:
@nicopace on every chef related website it say "chef is outdated". As well, I'm not intending to add a login system or public ssh key storage. Past users would wonder where to find these features. Third, chef.io exists.

nicopace, 2017-11-02:
@aparcar outdated doesn't mean that is dead.

The features that you don't plan to add may be added by someone else as optional features (plugins), and chef.io don't see why it is relevant.

On the other hand, chef is recognized by the AlterMundi's community, it is related with the 'cooking' term that lime-sdk is using, and it is understandable by everyone (chef is kind of a multilingual word).

Would vote for jefe as a replacement of chef and using it's name and domains.

aparcar, 2017-11-02:
@nicopace okay sound good to me. I'll shut down this repository and create a new one. Please create chef.libremesh.org
@ilario @FreifunkUFO are you two okay with this decision?

ilario, 2017-11-02:
I'm ok with the _chef_ name :)
The repository, I think you don't need to delete it, I think you can rename it, this way Github will redirect requests to the new one.

closed 2017-11-02 by aparcar

FreifunkUFO, 2017-11-03:
you are too fast when waiting for feedback

aparcar, 2017-11-03:
@FreifunkUFO sorry, hopefully you're okay with the decision of "chef"