libremesh/chef #19: qqq language files, translations via TranslateWiki and localization of web ui in the resulting firmware

ilario, 2017-11-12:
This discussion started on libremesh/alterchef#14 by @patogit I copy it here:

_Ok. The new Chef seems to be monolingual (English), with all the interface messages hard-coded in chef.html. Do you have any ideas about how we can internationalize this, in order to make easy translation possible? The first ideas that come to my mind are:_

_Find a way to replace the interface strings in the HTML file with keys that are rendered from a translation file, like LimeApp does at https://github.com/libremesh/lime-app/tree/uhttpd/i18n/translations. Then figure out when those pages get rendered, and how the user chooses their language._
_...or..._
_Create a parser-generator in PHP (this would run on TranslateWiki.net) that pulls out the interface strings from that file (making them available to translate on TW.net) and then writes the translated strings to a new files, such as chef-es.html for Spanish. Then we'd have to put an option somewhere on the interface to switch between languages._

_In the end, those two options are pretty similar to each other in terms of the work required. Maybe we can get better ideas from other software._

_Right now it has fewer than 10 strings, so another option is to just translate by hand for a few languages. That still leaves the question: How do we let the user know that there are multiple languages available, and switch between them?_

_Also, once Chef is available in Spanish, will it be able to cook firmware that gives my router an interface in Spanish?_

_In other words, I choose Spanish for the interface for Chef (or my browser automatically sends me to the Spanish version), I cook some firmware, I flash that firmware onto a router and reboot it, I login to the router, and the LibreMesh/LuCI interface is in Spanish... one seamless experience._

aparcar, 2017-11-12:
@ilario @patogit I'll check out the lime-app approach, I'm not up to write anything in PHP as it would require a different setup of the server, currently only Javascript.

The second point could be rather challenging, but chef can auto select a some additional packages. I'll look into that

patogit, 2017-11-12:
@aparcar the PHP parser-generator would only run on TranslateWiki.net. I've asked TW.net for examples of the parser-generators already in use, so that we can look at the code and think about how much effort it would require to write another. I've also asked them for ideas about how to internationalize the code. If you're curious, [here's the thread with them](https://translatewiki.net/wiki/Thread:Support/LibreRouter:_new_project_with_software_and_HowTo_booklets/reply_(9)).

aparcar, 2017-11-12:
@patogit thank you, keep me updated. I've never done any translations related stuff. We could split the task. I'll manage to build images in the selected language, you manage to translate the page?

aparcar, 2017-11-12:
@FreifunkUFO advises?

patogit, 2017-11-13:
@aparcar, I'm coordinating translation for the LibreRouter project, both software interfaces and documentation. I'm happy to take on the challenge of internationalizing the interface for Chef, but since it requires coding that I have little experience with, I'll look around for other people who have more coding skill than I do. @nicopace might have some ideas about this too.

As for building images in the selected language, how about this: Show a checkbox list of available languages, so that I can check the box next to each language I want, and (as a bonus feature) the language that I'm using for the Chef page is automatically selected. For example, where I live, it would be useful to have both English and Spanish. I imagine that a lot of communities might want English and their local language(s), since troubleshooting or asking for help online is sometimes easier in English.

Here are replies from TW.net:
> I've asked TW.net for examples of the parser-generators already in use, so that we can look at the code and think about how much effort it would require to write another:

_https://github.com/wikimedia/mediawiki-extensions-Translate/tree/master/ffs and docs in https://www.mediawiki.org/wiki/Help:Extension:Translate/File_format_support (likely somewhat outdated)._
Those parsers seem to range from about 150 lines of code up to 700 LOC.

> I've also asked them for ideas about how to internationalize the code:

_Maybe https://github.com/wikimedia/jquery.i18n ?_

I don't know how to implement that jQuery.i18n... but if I find some examples, I can probably figure it out. Or maybe it's too complicated for using on such a simple interface?

**Does jQuery.i18n look like a reasonable way to internationalize the new Chef? @nicopace, @nicoechaniz, @altergui, @gmarcos87, @p4u, do you have ideas about this?**

aparcar, 2017-11-14:
I'm currently not using jQuery in Chef but plain JavaScript. I'd rather keep it that way - if possible.

Some quick searches resulted in https://github.com/eligrey/l10n.js/ - maybe thats already enough?

Depending on the scope of Chef the complexity shouldn't increase at all or raise to a full config editor. That depends somewhat of what the community wants.

FreifunkUFO, 2017-11-14:
i think there is no need for a centralized, worldwide translation for this "chef":
there are less than 10 words to translate! 

atm no description and no help is existing, also the word "firmware" is not appearing anywhere, no description how to deal with the resulting "image" and some words to missunderstandings (f.i. image https://en.wikipedia.org/wiki/Image )

so i expect that mesh-communities will integrate this "chef" into own websites, with nice layout and pics, will write howtos and help-pages (f.i. how to flash firmware) and will translate it. 

localization of resulting firmware is another issue. but mostly done on openwrt/lede. sometimes the package-list have to be changed for the needed language.
f.i. packages like this: luci-i18n-base-de, luci-i18n-firewall-de,..

nicopace, 2017-11-14:
I understand your point of view @FreifunkUFO.
The good thing about making it translatable is that no matter if it has 1 or 10 words, someone that is not an english speaker will not be able to use the interface (imagine a first nation community from Mexico that only speaks Zapoteco).
We want to get to everyone... and for that, we need to have everything internationalized.

In relation to the rtanslation of the firmware, that is also part of the process that @patogit is taking care of (thanks Pato for getting involved in this).

@aparcar, @gmarcos87 has done this for the lime-app:
* https://github.com/libremesh/lime-app/pull/96
* https://github.com/libremesh/lime-app/pull/97

you can also ask him.

FreifunkUFO, 2017-11-29:
translation is in progress: https://github.com/libremesh/chef/blob/master/i10n/en.json

@aparcar is now asking: who wants to help with more languages/translations?

ilario, 2017-11-29:
italian: me @G10h4ck @gabri94 @digitigrafo @openspace42 
spanish: @altergui @gmarcos87 @nicoechaniz @nicopace @zenem @pablocastellano
catalan: @p4u @rogerpueyo 
german: @FreifunkUFO @aparcar 
portuguese: @brunovianna 
valencian: @amuuza 
anyone for other widely used languages? (French, Arabic, Mandarin, Russian, Hindi, Bengali, Japanese...)

ilario, 2017-11-29:
Italian: done

amuuza, 2017-11-29:
Happy to help.
Wow, even Valencian : o
I guess the more the better.
If it's ok with you, I will wait for @p4u or @rogerpueyo translation and then I will check if any change is needed for Valencian : )

patogit, 2018-02-13:
Maybe this week we can create a tw-translate branch and give direct push commit access for that branch to @translatewiki ... this will enable more people to translate via the TranslateWiki.net interface.

Any progress on making the firmware image relate to the chef interface language? (So that someone who chooses Spanish in the Chef interface and then flashes that image on their router gets a Spanish interface on LibreMesh on their router.)

patogit, 2018-02-16:
@aparcar, what branch do you want to receive the pull requests for translations from TranslateWiki? In other words, what's the development branch?

patogit, 2018-02-17:
@aparcar asked about a JS library supporting .po files. I found a few options, though PO is not a very modern format (there are other formats that can make programming easier) and I don't know enough about JS or chef to understand what library is the most appropriate.

**SIMPLER LIBRARIES**

1. https://github.com/messageformat/Jed
2. https://github.com/adrienrn/jsgettext
3. https://github.com/signavio/i18n
4. https://github.com/piranha/puttext/

**MORE COMPLEX LIBRARIES**

1. https://www.i18next.com/#
2. http://l10ns.org/
3. https://github.com/airbnb/polyglot.js (explained at https://stackoverflow.com/questions/35111121/multilingual-website-with-html/35111281#35111281 )

gmarcos87, 2018-02-18:
Hi @patogit , I create a pull request (https://github.com/libremesh/chef/pull/38) that incorporates the qqq.json file with the link to the images.
In this case it is always the same image because the chef has only one screen and it is very simple. 
@aparcar is already using a translation system, which loads the data from multiple json files (similar to lime-app). I don't think you need to change anything about that implementation, just add new languages or complete existing ones.

patogit, 2018-02-18:
@gmarcos87 great that you got a screenshot in there -- message documentation helps translation. Another option is to put a link to chef.libremesh.org in the message documentation so that a translator can go to the site, see the current version, and interact with it -- this might be even more useful than a link to a screenshot, since the screenshot might become obsolete as chef changes.

aparcar, 2018-03-21:
I think this can be closed due to @gmarcos87 work.

closed 2018-03-21 by aparcar