libremesh/chef #24: make some tickets to support freifunk

FreifunkUFO, 2017-11-29:
@aparcar was asking about supporting freifunk mesh firmwares. and their packagelists.
for these many freifunk networks we need many presets, packagelists, "distributions" (fi gluon and berlin are built on LEDE, but with big patch-lists)  and/or firmwares.
so we would need many tickets, at least one for each freifunk-approach

aparcar, 2017-11-29:
To support the different patched "distributions" the upgrade server need's all the ImageBuilders. We need to ask the FF-communities to create them as well and store them in the same structure as on official servers

@mwarning

FreifunkUFO, 2017-11-30:
at least some freifunk distributions are also working with standard-lede-imagebuilder.. but mostly there are patched. f.i. atm  freifunk-meshkit in leipzig has imagebuilder-builds with disabled ksymbols (to support common hardware, like cpe510).
you could try it with http://gadow.freifunk.net:8004/srv2/lede/lede-20171030/bin/targets/ar71xx/generic/lede-imagebuilder-SNAPSHOT-ar71xx-generic.Linux-x86_64.tar.xz

aparcar, 2017-12-23:
Specially patched imagebuilders are supported as long as they follow the official naming. 
An example can be found [here](https://chef.libremesh.org/#87467e0d4ab5607)

For snapshots is therefore important to disable the following two options:
```#CONFIG_VERSION_FILENAMES=y
# CONFIG_VERSION_CODE_FILENAMES=y
```

@mwarning @2tata if possible please provide patched imagebuilders in the official structure to enable custom image building via chef.

FreifunkUFO, 2017-12-28:
so, we would need at least one ticket for each freifunk-firmware-distribution? (weimar, berlin, leipzig, jena, bielefeld, gluon) details should be discussed there. 

we could start realization with firmware which is able to use unpatched openwrt/lede and then have a look about their configuration (which one are zeroconfig firmwares?). and ask other firmwares developers about their imagebuilder and details.

FreifunkUFO, 2017-12-28:
i started with 
https://github.com/libremesh/chef/issues/30
and 
https://github.com/libremesh/chef/issues/29

aparcar, 2017-12-28:
should we close this "meta issue" and focus on the specific ones?

FreifunkUFO, 2017-12-28:
ticket issue was to open tickets for freifunk. there are only 3 atm. what about f.i. @mwarning or weimar or jena @egon0 ?

mtippmann, 2018-01-03:
@FreifunkUFO maybe it's easier to implement a configuration file that contains all required community-specific urls so every community can setup their own instance of chef/update-server? In Weimar we'll probably need custom fields (nodenumber hack) and we also can't use the generic imagebuilder/sdk at the moment - so I don't think it's worthwhile to add us to the list at the moment.

egon0, 2018-01-07:
Jena will use plain libremesh with a network-profile soon, no need to do anything to support the old monkeypatched legacy Firmware.

aparcar, 2019-03-11:
Can Iclose this?

aparcar, 2021-03-14:
Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar