libremesh/chef #26: "Unknown package np-..." error for first compilation and no subsequent error

ilario, 2017-12-03:
This is a followup from the lime-users mailing list [here](https://lists.libremesh.org/pipermail/lime-users/2017-December/001022.html) for an issue described [here](https://wiki.bologna.ninux.org/mediawiki/index.php/Errore_profilo_LibreMesh_encrypt_nano) and [here](https://wiki.bologna.ninux.org/mediawiki/index.php/Errore_profilo_LibreMesh_encrypt).

I tried to reproduce and I also think that there's a bug, maybe in Chef or in repo.libremesh.org or some browser plugin that break Chef (but I confirmed with two different browsers)
In the log posted on lime-users there's an error, like if Chef could not reach http://repo.libremesh.org/network-profiles/:
`Unknown package 'np-libremesh-encrypt-11s'.`

Because of this error, a compilation for a new model (not one for which the image have already been cooked) selecting a non empty network profile fails often (not always, maybe just with new targets?). And all next builds fails pointing to exactly the same log as the first failed compilation.

Another strange thing is that some times selecting another profile seems solving but whichever network profile is selected, the image to download is identical.

aparcar, 2017-12-03:
I recently changed from network-profiles based on a folder to the network-profiles based on packages. That leads to images without the network-profile in name. To see if the package is installed please check the resulting `.manifest` in advanced view. 

Could you please write a specific device where the image appears? I could not reproduce the image.

brunovianna, 2017-12-03:
I tried to cook a tl-wa801nd-v1 with JaexaPora network profile, lime without web interace and got an error. Here is the link to the log
https://ledeupdate.planetexpress.cc/static/faillogs/request-b4e7c6e1ef26.log

aparcar, 2017-12-06:
@brunovianna  @ilario The issue was fixed in https://github.com/aparcar/attendedsysupgrade-server/commit/6276bf1ad8acc3b58a4fc24286840cba6ea4bfa1 please check again

closed 2017-12-17 by aparcar