libremesh/chef #28: impossible build image for ubiquiti loco xw

digitigrafo, 2017-12-17:
say: image building.

In console of browser (firefox 57):

chef.js:464:4
GETXHR
https://ledeupdate.planetexpress.cc/api/build-request/e83cb1e07ec8
[HTTP/1.1 202 ACCEPTED 45ms]
building
chef.js:464:4
GETXHR
https://ledeupdate.planetexpress.cc/api/build-request/e83cb1e07ec8
[HTTP/1.1 202 ACCEPTED 42ms]
....

for infinite time.
[image]

aparcar, 2017-12-17:
@digitigrafo I recently tried some new stuff and seems to cause some problems. I reset parts of the database and build the image, please try yourself.
https://chef.libremesh.org/#2478b9c37ce2b52

digitigrafo, 2017-12-23:
yes! thanks. now is working.

closed 2017-12-23 by digitigrafo