libremesh/chef #30: support freifunk leipzig

FreifunkUFO, 2017-12-28:
freifunk leipzig is using unpatched meshkit on unpatched (sometimes self-built) imagebuilder.
meshkit puts f.i. hostname and ip-adress into the firmware (not zeroconfig!) 
usage without meshkit (take lede, install freifunk-packages) also possible: then you have to use the luci-app-meshwizard for main (only olsr) config of the node
mesh-protocols: 2x batman-adv (on wifi-vlan 12 and 61) and olsr (on wifi, without vlan)

FreifunkUFO, 2017-12-28:
@aparcar startet tests on https://ledeupdate.planetexpress.cc/

in the build-log you can see the used buildroots ("ar71xx-lede-20171030") so this should also be selectable inside chef.

my first test gave me a firmware without any mesh routing protocol  (see https://ledeupdate.planetexpress.cc/json/ff-leipzig/snapshot/ar71xx/generic/tl-wdr3600-v1/vanilla/build-8462e4f5f4163d2.log )

the distribution name (ff-leipzig) should be renamed to freifunk-leipzig (or freifunk-meshkit-l). ff is not a common shortcut for freifunk.

FreifunkUFO, 2017-12-28:
meshkit is here: https://github.com/freifunk/meshkit

the community cities (profiles) are here: https://github.com/openwrt/luci/tree/master/contrib/package/community-profiles/files/etc/config

and local community files (f.i. webui-headerimage and postinstall-scripts) are here: https://github.com/freifunk/meshkit-community_files

FreifunkUFO, 2018-02-20:
with freifunk-meshkit these community files are given to the imagebuilder-process by parameter "FILES". that should be supported by image server and chef. 
thats also the current way to transport node-specific settings from the imagebuilder-webinterface (meshkit) to the device (f.i. /etc/config/freifunk with Mesh- IPv4 address and LAN/WAN/WLAN-modifications). when flashing there is a package "meshwizard", it is reading that config-file and executes all preconfigured stuff (f.i. ipv4 mesh adress)
 
unfortunatly @aparcar was closing a similiar ticket https://github.com/libremesh/chef/issues/34 (exceeds the scope), because on libremesh these community files have to be inside community opkg-packages.

aparcar, 2018-02-20:
@FreifunkUFO could you consider using packages as well? Offering raw files will lead to rebuilding equal images which is out of the scope of the image builder. The idea is to cache created images

FreifunkUFO, 2018-12-14:
yes, that would lead to build individual images for each device. that should be on scope of the image builder.
meanwhile, i saw an option "edit uci defaults" on a beta-chef-instance. so this feature is now on preparation?

aparcar, 2019-01-11:
> meanwhile, i saw an option "edit uci defaults" on a beta-chef-instance. so this feature is now on preparation?

Yes, feel free to try it. Individual images are deleted within 7 days and "hidden", so nobody with exactly the same configuration can download you image

aparcar, 2019-01-16:
the following could do the trick. On first boot, the file should be created, followed by a reboot causing meshwizard to (re)load the config. I don't know how meshwizard works, however, based on the example below, it is possible to create files via Chef.

```
#!/bin/sh

cat <<EOF >>/etc/config/meshwizard
config system 'system'
option hostname '120-30'
option conloglevel '8'
option cronloglevel '9'
option latitude '51.33348'
option longitude '12.40297'
option location ''

config public 'community'
option name 'leipzig'

config public 'contact'
option nickname ''
option name ''
list homepage ''
option mail ''
option phone ''
option note ''

config core 'luci_main'
option mediaurlbase '/luci-static/freifunk-generic'

config netconfig 'netconfig'
option IB_wifi0_config '1'
option IB_wifi0_ip4addr '10.61.120.30'
option IB_wifi0_channel '1'
option IB_wifi0_vap '1'

config netconfig 'wan'
option proto 'dhcp'

config netconfig 'lan'
option proto 'static'
option ip4addr '192.168.1.1'
option netmask '255.255.255.0'

config defaults 'ipv6'
option enabled '0'

config general 'general'
option sharenet '0'
option localrestrict '1'
EOF

reboot
```

FreifunkUFO, 2019-01-21:
ok thank you for explanation

i didnt found docu, but the file which will be created: /etc/uci-defaults/99-server-defaults 
i doubt using the reboot-command at this moment. (see https://openwrt.org/docs/guide-developer/uci-defaults )

the pull-request is here https://github.com/aparcar/attendedsysupgrade-server/pull/115/files

FreifunkUFO, 2019-01-21:
i there is no need for the first two lines. (/bin/sh) 

f.i. on freifunk-nodes i see small scripts like:
cat /rom/etc/uci-defaults/freifunk
uci set uhttpd.main.rfc1918_filter=0
uci commit uhttpd

aparcar, 2019-03-11:
@FreifunkUFO whats missing here?

aparcar, 2021-03-14:
Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar