libremesh/chef #31: support freifunk berlin

FreifunkUFO, 2017-12-28:
berlin is using self-build firmware with large patchsets, own openwrt/lede-packages and own webui-wizard for configuration.

see https://github.com/freifunk-berlin/firmware and packages are at https://github.com/freifunk-berlin/firmware-packages

aparcar, 2017-12-28:
@andrenarchy ping

andrenarchy, 2017-12-31:
Also looping in @SvenRoederer and @booo from Berlin. :)

aparcar, 2018-03-21:
@andrenarchy could you please provide the imagebuilders of your current release?

SvenRoederer, 2018-03-22:
they can be found in https://buildbot.berlin.freifunk.net/buildbot/stable/1.0.0/

aparcar, 2018-09-06:
Hi @SvenRoederer , do you mind changing the folder structure to the official format of `target/subtarget` instead of `target-subtarget`? Then it'd be compatible which the server backend

aparcar, 2021-03-14:
Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar