libremesh/chef #32: support freifunk bielefeld

FreifunkUFO, 2017-12-29:
bielefeld is using an own freifunk-firmware (of course openwrt/lede)
https://github.com/freifunk-bielefeld/firmware
some special packages.. two patchset-directories..

FreifunkUFO, 2017-12-29:
ping @mwarning, what do you think. would it be possible?

https://github.com/libremesh/chef/issues/24

mwarning, 2018-01-07:
not sure, I have to try. I was sick over last week.

aparcar, 2018-03-21:
@mwarning could you please provide the imagebuilders of your current release?

aparcar, 2018-08-20:
@mwarning ping

mwarning, 2018-08-20:
@aparcar we still struggle to get the next release ready for the release..

aparcar, 2018-09-06:
@mwarning any updates?

mwarning, 2018-09-06:
this weekend

aparcar, 2018-09-06:
Please follow the official folder structure `[version]/targets/[target]/[subtarget]/` if possible :+1:

mwarning, 2018-10-24:
ok, we finally had out 1.0.0 release. Let's see what needs to be done. :-)
@aparcar is the structure for the images? We have them all in one folder.

aparcar, 2018-10-24:
I need the image builder in the previously mentioned structure for all targets you'd like to support

mwarning, 2018-10-26:
Would this do? http://firmware.freifunk-bielefeld.de/freifunk/firmware/1.0.0/targets/

aparcar, 2018-10-27:
Perfekt!

aparcar, 2018-10-30:
@mwarning actually sorry, the structure is not perfect yet. The [meta-imagebuilder](https://github.com/aparcar/meta-imagebuilder/) validates downloaded imagebuilders via checksums and [gpg](https://github.com/aparcar/meta-imagebuilder/blob/master/meta#L64). I could disable it specifically for FF Bielefeld, but a nicer solution would be you sign the created imagebuilders as well

mwarning, 2018-10-30:
hm, what do I have to do? What gpg key has to be used?

aparcar, 2018-10-30:
Nevermind, I'll use https and adapt the meta imagebuilder :)

mwarning, 2018-10-30:
Fine with lazy myself. :D

aparcar, 2021-03-14:
Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar