libremesh/chef #34: easy way to add custom /etc/config files

pedro-nonfree, 2018-01-14:
Would like the ability to upload custom /etc/config files to be included in the resulting firmware binary image

aparcar, 2018-01-15:
Yes. Create a network profile and merge it into the official libre mesh repo. The server will build packages put of it which are installable. 

A different approach to allow per image different settings exceeds the current scope of the image server. Hard to manage versions and outdated images

ilario, 2018-01-15:
You can take as an example the other network-profiles, they all have files in /etc/config, doesn matter if it's lime-defaults or another file.

aparcar, 2018-01-15:
the created packages are [here](http://repo.libremesh.org/network-profiles/), you could even install multible of them with different ssh-keys or whatever you need

pedro-nonfree, 2018-02-18:
Thanks for your responses!

closed 2018-02-18 by pedro-nonfree