libremesh/chef #54: icon for language selector

patogit, 2018-05-26:
Putting a language selector icon, like the one on www.wikipedia.org... but that one is in a sprite, and I can't find the file.

patogit, 2018-06-10:
here are two links to the file: [first](data:image/svg+xml,%3Csvg%20xmlns=%22http://www.w3.org/2000/svg%22%20width=%2218%22%20height=%2218%22%20viewBox=%220%200%2024%2024%22%3E%20%3Cpath%20fill=%22%2372777d%22%20d=%22M13%2019l.8-3h5.3l.9%203h2.2L18%206h-3l-4.2%2013H13zm3.5-11l2%206h-4l2-6zM5%204l.938%201.906H1V8h1.594C3.194%209.8%204%2011.206%205%2012.406c-1.1.7-4.313%201.781-4.313%201.781L2%2016s3.487-1.387%204.688-2.188c1%20.7%202.319%201.188%203.719%201.688l.594-2c-1-.3-1.988-.688-2.688-1.188%201.1-1.1%201.9-2.506%202.5-4.406h2.188l.5-2H7.938L7%204H5zm-.188%204h3.781c-.4%201.3-.906%202-1.906%203-1.1-1-1.475-1.7-1.875-3z%22/%3E%20%3C/svg%3E) and [second](data:image/svg+xml,%3Csvg%20xmlns=%22http://www.w3.org/2000/svg%22%20width=%2220%22%20height=%2220%22%3E%20%3Cpath%20d=%22M20%2018h-1.44a.61.61%200%200%201-.4-.12.81.81%200%200%201-.23-.31L17%2015h-5l-1%202.54a.77.77%200%200%201-.22.3.59.59%200%200%201-.4.14H9l4.55-11.47h1.89zm-3.53-4.31L14.89%209.5a11.62%2011.62%200%200%201-.39-1.24q-.09.37-.19.69l-.19.56-1.58%204.19zm-6.3-1.58a13.43%2013.43%200%200%201-2.91-1.41%2011.46%2011.46%200%200%200%202.81-5.37H12V4H7.31a4%204%200%200%200-.2-.56C6.87%202.79%206.6%202%206.6%202l-1.47.5s.4.89.6%201.5H0v1.33h2.15A11.23%2011.23%200%200%200%205%2010.7a17.19%2017.19%200%200%201-5%202.1q.56.82.87%201.38a23.28%2023.28%200%200%200%205.22-2.51%2015.64%2015.64%200%200%200%203.56%201.77zM3.63%205.33h4.91a8.11%208.11%200%200%201-2.45%204.45%209.11%209.11%200%200%201-2.46-4.45z%22%20fill=%22%2354595d%22/%3E%20%3C/svg%3E)

aparcar, 2018-08-20:
Here is [a icon set](http://www.famfamfam.com/lab/icons/flags/) you might want to use.

aparcar, 2018-09-06:
There is work happening on a chef rewrite in vuejs, please check this out and reopen the issue there:
https://github.com/edvm/chef

patogit, 2018-10-18:
> There is work happening on a chef rewrite in vuejs, please check this out and reopen the issue there:
> https://github.com/edvm/chef

@aparcar GitHub does not allow me to open an issue in that repo.

the icon I want to add to the language selector is available in SVG and PNG:
https://github.com/wikimedia/mediawiki-extensions-UniversalLanguageSelector/blob/master/resources/images/compact-links-trigger.svg
https://github.com/wikimedia/mediawiki-extensions-UniversalLanguageSelector/blob/master/resources/images/compact-links-trigger.png

aparcar, 2018-10-19:
> > There is work happening on a chef rewrite in vuejs, please check this out and reopen the issue there:
> > https://github.com/edvm/chef
> 
> @aparcar GitHub does not allow me to open an issue in that repo.
The devs seems to stopped development (for now), so no worries requesting it here.

> the icon I want to add to the language selector is available in SVG and PNG: [...]

I don't fully understand, you want to add a single icon or a flag icon per language? In both cases please check the HTML code in chef.html

patogit, 2019-03-23:
[this is the image I want to add](https://nube.altermundi.net/index.php/s/Q6LAQGkPrL3xaGG). I want it to appear to the left of the language selector menu, so I'm guessing there are two steps: put the image file in the repo, and then add a line after line 16 in chef.html that puts the image in the correct place with the correct alignment.

patogit, 2019-03-25:
What's the development branch? In other words, if I make the change required and make a pull request, what branch shall I send the request to?

aparcar, 2019-03-25:
always master

aparcar, 2021-03-14:
We got some icons for the language selector in the new interface.

Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar