libremesh/chef #59: Support of disro/version descriptions and aliases

aparcar, 2018-08-01:
The new API offers some further information about versions and distributions, see an example below. Both should be included (if available) in the new chef design to help the user

https://as-test.stephen304.com/api/versions
https://as-test.stephen304.com/api/distros

edvm, 2018-08-06:
@aparcar working on this :+1:

aparcar, 2018-08-06:
The current API implementation returns an array, I could change that to return a dict like this
```
{
    "openwrt": { "alias": OpenWrt", "latest": "18.06.0" },
    "lime": { "alias": "LibreMesh", "latest": "17.06" }
}
```

edvm, 2018-08-06:
@aparcar its fine that the API returns a list of elements, please dont change it :)

aparcar, 2018-09-06:
The [current dev version of Chef](https://as-test.stephen304.com/chef/) already supports this,
[image]

closed 2018-09-06 by aparcar