libremesh/chef #62: dnsmasq-full

pozzo-balbi, 2018-09-06:
Hi, selecting dnsmasq-full to use mwan3 leads to the following error:

Collected errors:
 * check_data_file_clashes: Package dnsmasq wants to install file /mnt/worker/worker1/imagebuilder/lede/17.01.5/ar71xx/generic/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/hotplug.d/ntp/25-dnsmasqsec
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /mnt/worker/worker1/imagebuilder/lede/17.01.5/ar71xx/generic/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/init.d/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /mnt/worker/worker1/imagebuilder/lede/17.01.5/ar71xx/generic/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/usr/sbin/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * opkg_install_cmd: Cannot install package dnsmasq.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

https://chef.libremesh.org/ is not having this behavior. Thanks

aparcar, 2018-09-06:
Hi @pozzo-balbi 
the problem occurs as the *old* chef creates a diff between packages in the *package text area* and the default packages, while the *new* chef behaves like the official ImageBuilder, removing only packages prefixed with a `-`.

It is possible to implement the diff behavior in the new chef as well, however I though of sticking to the official one for compatibility. 

An example, in *old* chef a package list only containing the package `htop` would remove **all** other packages an only install `htop`, while the new *chef* would install all default packages and add `htop` as an additional package.

@dangowrt whats your opinion on that?

pozzo-balbi, 2018-09-06:
Yes, great!! It works!! So in the old chef I would delete "dnsmasq" and add "dnsmasq-full". In the development version I put it as follows:

> -dnsmasq
> dnsmasq-full

and yes I can confirm it works. So maybe just add a note to this new behavior. Thanks so much for this great tool!!

closed 2018-09-06 by pozzo-balbi