libremesh/chef #64: Build issues

1gf948c2llew, 2018-09-22:
Says building but never finishes. Also WRT32X snapshot build manifest failures previous to this that, that don't clear after 24hrs for  / mvebu / cortexa9 / .

Thanks.

1gf948c2llew, 2018-09-23:
https://as-test.stephen304.com/download/faillogs/faillog-96e8f4729eba.txt

1gf948c2llew, 2018-10-03:
https://as-test.stephen304.com/download/faillogs/faillog-8b81e8b31f12.txt
https://as-test.stephen304.com/download/faillogs/faillog-b7bba45f737e.txt

aparcar, 2018-10-12:
I have to investigate a bit more as it seems to be a general problem for snapshots...
Thanks for testing and please be patient

1gf948c2llew, 2018-10-13:
okidoke, thanks for the update.

1gf948c2llew, 2018-10-18:
Still have issues after your update on openwrt forum. Thanks for trying though,
https://as-test.stephen304.com/download/faillogs/faillog-8b81e8b31f12.txt

aparcar, 2018-10-19:
Looks like `wpad-mini` was renamed to `wpad-mini`?

```
Collected errors:
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/sbin/hostapd
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/sbin/wpa_supplicant
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/sbin/wpad
	But that file is already provided by package  * wpad-mini
 * opkg_install_cmd: Cannot install package wpad-basic.
```
I'm on it!

1gf948c2llew, 2018-10-19:
Like a car bonnet!! :) Brilliant, Thanks v much bud.

Roland-B, 2018-11-21:
Any progress? Looking forward to being able to gen snapshots again..

aparcar, 2018-12-12:
@Roland-B please try again

aparcar, 2019-01-21:
@Roland-B the unstable is added as you're using a snapshot. These are nightly builds with may fail to start or work at all. Therefore don't use them if you depend on them. It is not related to the package selection itself.

closed 2019-01-21 by aparcar