libremesh/chef #66: There is 'lime_mini' but no 'lime_default' for the mt7628  for my GL-Inet GL-MT300N V2 's

Fibula1, 2018-10-21:
Hi all, There is 'lime_mini' but nothing under 'lime_default' for the mt7628 . I have a few GL-Inet GL-MT300N V2 and the mini wont let me change the password or give access to any setup menus so i'd like to try the 'lime_default' . Id love to get a mesh up and kicking on my units.
The GL-Inet GL-MT300N V2 is not available in the chef builder or developer chef builder page. Maybe one of you kind folk could help build a 'lime_default' for my MT300N V2 and put it in the folder ?

Thanks in advance, Phil.

http://repo.libremesh.org/dayboot_rely/17.06/targets/ramips/mt7628/mt7628/lime_mini/
http://repo.libremesh.org/dayboot_rely/17.06/targets/ramips/mt7628/mt7628/lime_default/

aparcar, 2018-10-22:
The router should have enough storage to handle the default build as well. I'll look into that later why chef fails to show and build the firmware

Fibula1, 2018-10-25:
> The router should have enough storage to handle the default build as well. I'll look into that later why chef fails to show and build the firmware

Hi Aparcar, any more thoughts? Thx

Fibula1, 2018-12-13:
Hi Aparcar, any more thoughts? Thx  (ps i havent tried chef-builder again)

aparcar, 2019-02-01:
Hi @Fibula1 sorry for the late reponse. The stuff from repo.libremesh.org is not really related to chef, I'm not managing these. However which Chef you can create any image you like, see [here](https://chef.libremesh.org/#62fc8cc00fe953b)

You can choose from some pre defined LibreMesh flavors or customize it yourself.

Please close this issue if it works for you.

closed 2019-03-11 by aparcar