libremesh/chef #67: translations repo out of sync

patogit, 2018-10-29:
when I create a new pull request for new translation commits, GitHub tries to pull in all the commits ever made in the translation repo, instead of just the new commits. I don't know if rebasing is the appropriate action. Please suggest how we can fix this. @aparcar and @gmarcos87 maybe you have ideas.

patogit, 2018-10-29:
maybe this, on my computer:
1. cd ~/Documents/GitHub-translation-bridge
2. sudo rm -r chef
3. git clone https://github.com/translation-bridge/chef.git
4. cd chef
5. git remote add upstream https://github.com/libremesh/chef.git
6. git status (to make sure I'm on master branch)
7. git fetch upstream
8. git rebase upstream/master
9. git push origin

Would that work? Or would it erase the recent Oct 22 commit of new translations https://github.com/translation-bridge/chef/commits/master ?

aparcar, 2018-12-12:
Hey @patogit I merged the translation myself, should work now.

closed 2018-12-12 by aparcar