libremesh/chef #70: Cannot build any LibreMesh firmware for version 17.06 and up

markbirss, 2018-11-28:
17.06 - Unknown package in request: kmod-ipt-offload
18.06.1 - Unknown package in request: libremesh-default
18.06.1-dev - Unknown package in request: libremesh-default

ilario, 2018-11-28:
I suppose you're using the development Chef server?
If so, please fill the issue on chef repository https://github.com/libremesh/chef or maybe @aparcar can comment here?
I just tried the stable Chef on https://chef.libremesh.org and it works well.

markbirss, 2018-11-28:
Yes issue is with beta chef.  Original chef only builds libremesh up to
version  17.06 not anything with more recent openwrt base

aparcar, 2018-12-11:
@markbirss please try again now, some recent changed should fix it

markbirss, 2018-12-11:
Hi Paul

Thank you for your reply, the build starts, however the following error
with LibreMesh version

*18.06.1*
*https://as-test.stephen304.com/download/faillogs/faillog-cc996b3d1037.txt
<https://as-test.stephen304.com/download/faillogs/faillog-cc996b3d1037.txt>*

Collected errors:
 * check_data_file_clashes: Package libremesh-default wants to install
file /mnt/worker/worker2/imagebuilder/lime/18.06.1/ramips/mt7620/build_dir/target-mipsel_24kc_musl/root-ramips/etc/config/lime-defaults
	But that file is already provided by package  * lime-system
 * opkg_install_cmd: Cannot install package libremesh-default.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

*18.06.1-dev*
*https://as-test.stephen304.com/download/faillogs/faillog-11d8aaaeddc6.txt
<https://as-test.stephen304.com/download/faillogs/faillog-11d8aaaeddc6.txt>*

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies
for lime-full:
 * 	lime-basic * 	luci-app-lime-location *
 * opkg_install_cmd: Cannot install package lime-full.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

Regards

aparcar, 2018-12-11:
Try 18.06.1-dev

markbirss, 2018-12-12:
Hi

The page at

https://chef.libremesh.org/

has error

not allow to choose Distribution at all the list of options is currently
empty ?

Regards

aparcar, 2018-12-12:
Sorry the server was down for the last hours

v4l3r10, 2018-12-21:
Hi,

i'm trying to build an image but something not work:
https://as-test.stephen304.com/download/faillogs/faillog-3e8df0220a67.txt

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-full:
 * 	lime-basic * 	luci-app-lime-location * 
 * opkg_install_cmd: Cannot install package lime-full.

aparcar, 2018-12-21:
Please try again. snapshots.libremesh.org was down and therefore some packages missing. Should work again

FreifunkUFO, 2019-01-11:
Where is a beta-chef or even another chef (which is not chef.libremesh.org)?

aparcar, 2019-01-11:
@FreifunkUFO for now there is only chef.libremesh.org. It kinda never left beta :)

FreifunkUFO, 2019-01-11:
ok, best would be to have three systems: stable, testing and development
please provide URLs for these.. where can i build libremesh with openwrt 18.xx ?

FreifunkUFO, 2019-01-16:
ok, there are some changes in https://github.com/aparcar/attendedsysupgrade-server/tree/master/distributions/lime so there is now a libremesh with openwrt-18.xx

i wonder about reading "17.xx" in  https://github.com/aparcar/attendedsysupgrade-server/blob/master/distributions/lime/testing.yml#L14 (on libremap and limeui repo, but maybe both are outdated anyway?)

markbirss, 2019-01-17:
> @FreifunkUFO <https://github.com/FreifunkUFO> I'm afraid both are legacy,
> maybe @p4u <https://github.com/p4u> can say something about that as he
> setup the repo

I have since instead been building my own firmware completely from source,
as i have also some devices which don't show up on chef.libremesh.org

like Cisco Meraki MR24 and others

Regards

aparcar, 2019-01-17:
you can find here a list of active targets, if you want more, please create a PR
https://github.com/aparcar/attendedsysupgrade-server/blob/master/distributions/lime/distro_config.yml#L9

closed 2019-01-21 by aparcar