libremesh/chef #71: Add INSTALL documentation

FreifunkUFO, 2018-12-14:
please provide documentation howto install and host such a system.

i think, at least there is a need for changing the "server" in https://github.com/libremesh/chef/blob/master/chef.html#L11 (when having an own one)

corresponding ticket for the attendedsysupgrade-server (also needed) is https://github.com/aparcar/attendedsysupgrade-server/issues/135

aparcar, 2018-12-19:
The installation is pretty straight forward, just put the files somewhere on a server and point a web server to open the `chef.html` file. For further *settings* please see [this line](https://github.com/libremesh/chef/blob/master/chef.html#L11). Basically you just have to set a working backend.

FreifunkUFO, 2019-01-11:
ok, "this line" on chef.html was already linked in my ticket/issue-description.

atm, the problem is setting up the attendedsysupgrade-server on the same/ or another host.. 

see also https://github.com/aparcar/attendedsysupgrade-server/issues/138 (and 135)

aparcar, 2019-01-11:
So let's keep the discussion in one issue. If setting up Chef (without the backend) isn't the problem please close this issue...

FreifunkUFO, 2019-01-21:
chef without attendedsysupgrade-server isnt that useful. 
please add some information about attendedsysupgrade-server on the chef-github. (not only that existing one small phrase).

other INSTALL problem might be: there is only a chef.html but no index.html .. how is it realized to run https://chef.libremesh.org/ without the chef.html?

aparcar, 2019-01-21:
https://github.com/libremesh/chef/commit/fd696e52f7be5ee13bdb9d0e02e13105c8149581

closed 2019-01-21 by aparcar