libremesh/chef #73: You're not moving forward in queue for a long time

Drdoomblunt, 2019-01-03:
Have been trying to make a slimmer firmware for the wr940n v6, using OpenWRT 18.0.6.1

I have tried on 3 occasions, after reboots and with seperate browsers to try eliminating any issues my end.

I am trying to build a web interface version with the following packages:

base-files
busybox
dnsmasq
dropbear
firewall
fstools
iptables
kmod-ath9k
kmod-gpio-button-hotplug
libc
libgcc
logd
luci
luci-app-openvpn
mtd
netifd
openvpn-openss1
opkg
swconfig
uboot-envtools
uci
uclient-fetch
wpad-mini

DittelHome, 2019-01-03:
Same here:
You're not moving forward in queue for a long time, please report this
You are in the build queue. Position: 110

Drdoomblunt, 2019-01-03:
I am now 116. Is the build queue itself stuck on something?

Drdoomblunt, 2019-01-08:
See https://github.com/aparcar/attendedsysupgrade-server/issues/136, build queue now working.

closed 2019-01-08 by Drdoomblunt