libremesh/chef #74: chef libremesh raspberry pi 3

leocruz0, 2019-01-16:
Hi folks. I am having a problem to download the image for my raspberry pi 3 on chief. that is the error log:
uilding package index...
Downloading http://downloads.openwrt.org/releases/18.06.1/targets/brcm2708/bcm2709/packages/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_core
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_base
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_routing
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_luci
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_packages
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/reboot_telephony
Downloading file:packages/Packages
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/imagebuilder
Downloading https://snapshots.libremesh.org/packages//Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/libremesh
Downloading http://repo.libremesh.org/releases/17.06/packages/arm_cortex-a7_neon-vfpv4/libremap/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/libremap
Downloading http://repo.libremesh.org/releases/17.06/packages/arm_cortex-a7_neon-vfpv4/limeui/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/limeui
Downloading https://as-test.stephen304.com/network-profiles//Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/lime/testing/brcm2708/bcm2709/dl/lm_profiles
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-full:
 * 	lime-basic * 	luci-app-lime-location * 
 * opkg_install_cmd: Cannot install package lime-full.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

aparcar, 2019-02-01:
Should be fixed via https://github.com/libremesh/lime-packages/pull/462

aparcar, 2019-02-01:
works now https://chef.libremesh.org/#bf2c479309cbf44

closed 2019-02-01 by aparcar