libremesh/chef #79: OM2P/OM5P-AC sysupgrade.bin or factory.bin image missing

wifilogo, 2019-02-01:
Hi!
I'm trying to generate two images for OM2P and OM5P-AC (OpenWrt 18.06.1 with web interface) but not sysupgrade.bin or factory.bin images are generated.

`
Installed packages (93)
Created files (9)

    buildlog-73adf15b6e756bf.txt
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-device-om5pac.manifest
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-root.squashfs
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-uImage-lzma.bin
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-vmlinux-lzma.elf
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-vmlinux.bin
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-vmlinux.elf
    openwrt-18.06.1-d533f9491319950-ar71xx-generic-vmlinux.lzma
    sha256sums

`
Can you help me, please?
Congrats for your excellent work!

aparcar, 2019-02-01:
could you try to build the image locally? with success? 
I trier 17.02, 18.06.{0,1,2} and snapshots, but it's never created.

wifilogo, 2019-02-01:
Yes. I compile the image correctly with local Image Builder.

`
ls -l /home/ib_ar71xx/bin/targets/ar71xx/generic
total 45232
-rw-r--r-- 1 root root    3362 Dec 30 13:05 openwrt-18.06.1-ar71xx-generic-device-gl-ar300m.manifest
-rw-r--r-- 1 root root    3380 Dec 25 21:15 openwrt-18.06.1-ar71xx-generic-device-om2p.manifest
-rw-r--r-- 1 root root    3465 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-device-om5pac.manifest
-rw-r--r-- 1 root root 5242884 Dec 30 13:05 openwrt-18.06.1-ar71xx-generic-gl-ar300m-squashfs-sysupgrade.bin
-rw-r--r-- 1 root root 5627805 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-om5pac-squashfs-factory.bin
-rw-r--r-- 1 root root 5627805 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-om5pac-squashfs-sysupgrade.bin
-rw-r--r-- 1 root root 4325376 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-root.squashfs
-rw-r--r-- 1 root root 1367322 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-uImage-lzma.bin
-rwxr-xr-x 1 root root 1377444 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-vmlinux-lzma.elf
-rwxr-xr-x 1 root root 4355756 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-vmlinux.bin
-rwxr-xr-x 1 root root 4360880 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-vmlinux.elf
-rw-r--r-- 1 root root 1376256 Feb  1 11:35 openwrt-18.06.1-ar71xx-generic-vmlinux.lzma
-rw-r--r-- 1 root root    1800 Feb  1 11:35 sha256sums
-rw-r--r-- 1 root root 5242884 Dec 28 13:13 sp_openwrt-18.06.1-ar71xx-generic-gl-ar300m-squashfs-sysupgrade.bin
-rw-r--r-- 1 root root 5365662 Dec 25 21:15 sp_openwrt-18.06.1-ar71xx-generic-om2p-squashfs-factory.bin
-rw-r--r-- 1 root root 5365662 Dec 25 21:15 sp_openwrt-18.06.1-ar71xx-generic-om2p-squashfs-sysupgrade.bin

`

aparcar, 2019-02-01:
Hey the problem is the usage of BIN_DIR, which is done by the server. I reported this problem upstream, they have to fix it...

aparcar, 2019-02-12:
fix comes to snapshots soon https://github.com/openwrt/openwrt/pull/1815

closed 2019-03-11 by aparcar