libremesh/chef #80: chef.libremesh.org fails giving errors

h00man, 2019-02-16:
Hello, if I'm not mistaken, it seems like chef is cooking problems

-----------------------------------
Updated list of available packages in
/home/aparcar/worker/imagebuilder/lime/testing/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/../../../../../../../../../../../home/aparcar/worker/imagebuilder/lime/testing/ar71xx/generic/dl/lm_profiles
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies
for lime-full:
 * libremap-agent
 * opkg_install_cmd: Cannot install package lime-full.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
-----------------------------------

I went to https://chef.libremesh.org/
and tried to cook with options
distribution - libremesh
version - testing
router - tp-link wdr4300 v1
flavor - full libremesh
profile - libremesh default

aparcar, 2019-02-16:
@p4u you reported the same issue right?

aparcar, 2019-02-16:
Fixed via https://github.com/libremesh/lime-packages/pull/471

closed 2019-03-11 by aparcar