libremesh/chef #82: I think I broke zyxel builds..

sokoow, 2019-03-06:
Hey, I tried to build zyxel 2812 F3 couple of times and now I'm only getting this: https://chef.libremesh.org/download/faillogs/faillog-b88de8aa7419.txt

```
verifying sha256sums signature
gpg: assuming signed data in 'sha256sums'
gpg: Signature made Wed 30 Jan 2019 11:10:59 PM PST
gpg:                using RSA key 0F20257417E1CE16
gpg: Good signature from "OpenWrt Release Builder (18.06 Signing Key) <openwrt-devel@lists.openwrt.org>" [ultimate]
verified sha256sums signature.
ln: failed to create symbolic link '/home/aparcar/worker/imagebuilder/openwrt/18.06.2/lantiq/xrx200/key-build': File exists
```

it stopped being nice to me - help ? :D

aparcar, 2019-03-06:
Should be fixed via https://github.com/aparcar/meta-imagebuilder/commit/d4251ecac38ac7fa448233073aaa3f44cee42260

sokoow, 2019-03-06:
Cool, that is working now, but some builds started throwing me ```Failed to load resource: the server responded with a status of 409 (CONFLICT)```

aparcar, 2019-03-11:
Means that packages are in conflict, like `odhcpd-ipv6only` and `odhcpd`. Check the buildlog to see what went wrong.

closed 2019-08-17 by aparcar