libremesh/chef #84: [Feature] Support LibreCMC

ghbjklhv, 2019-06-23:
**Feature**: Support [LibreCMC](https://www.wikipedia.org/wiki/LibreCMC)
**Why?**: LibreCMC is the most popular router operating system for free software devices, especially RYF certified devices. I think if LibreMesh could be compiled with LibeCMC it would make is significantly easier for free software projects to support LibreMesh.
**How?**: LibreCMC is actually very similar to OpenWRT. My understanding is that they often collaborate on software projects because of this.

_________________________

Wan to support this issue? Post a bounty on it! We accept bounties via [BountySource](https://www.wikipedia.org/wiki/Bountysource).

[image]

ilario, 2019-06-23:
That's interesting but we would need to know more details about "very similar"/the small differences.
Do you have some list of differences?
On a first look seems that LibreCMC also uses UCI, and this is good.

aparcar, 2019-06-23:
It looks like a rebranding of OpenWrt, I did not find a description point out the difference. Feel free to create a PR. ImageBuilders are required for support.

ghbjklhv, 2019-06-24:
@ilario I'm unaware of such a list.

The key difference is that LibreCMC uses `[linux-libre](https://gogs.librecmc.org/libreCMC/linux-libre)` and `[linux-deblob](https://gogs.librecmc.org/libreCMC/linux-deblob)` scripts.
My understanding is that OpenWRT uses vanilla Linux.

Their source code is published here: https://gogs.librecmc.org/libreCMC/libreCMC

According to their FAQ:

> What is libreCMC based upon?
> 
> libreCMC GNU/linux-libre is a fork of the popular LEDE Project, but we use a linux-libre kernel and strip out some non-free components that are otherwise included in LEDE.
> What is wrong with OpenWRT/LEDE?
> 
> Some people have the view that everyone should have the freedom to control the software that runs on devices they own. If any compromises are made, then some of the benefits of Free Software go out the window. The OpenWRT/LEDE project does help give users more freedom than they would otherwise have, but some of the supported devices and packages are non-free.

aparcar, 2019-06-25:
I'll look into that next week

aparcar, 2019-07-02:
@ghbjklhv could you ask the developer to work more closely together with upstream OpenWrt? Would be interesting to apply his kernel patchset etc to the currently running OpenWrt stack.

As a base, ImageBuilders are required to use Chef, without that, support is currently a no go.

aparcar, 2021-03-14:
Closing due to lack of response. 

Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar