libremesh/chef #85: help!

chancat87, 2019-11-30:
Is the project still running? Why can't I open each of your websites?

Option cannot be selected, language cannot be switched?
[image]

aparcar, 2019-11-30:
Yes i got multiple requests for a new setup already. I'll try to fix it within the next week

chancat87, 2019-12-01:
> 是的，我已经有多个要求进行新设置的请求。我会在下周内尝试修复

Thank you for reply.

aparcar, 2020-03-09:
Uh oh sorry for the slow response! Please check the new front end at chef.libremesh.org/

FreifunkUFO, 2020-06-08:
error 
`AssertionError: Wrong ImageBuilder archive checksum`
on that web-page
linked to a new try of project 
https://github.com/mwarning/yet_another_firmware_selector

aparcar, 2021-03-14:
Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar