libremesh/chef #86: Lost the ability to remove packages from the default set.

TLCary, 2020-03-09:
Customize text box with LuCi...  and that's it.  So I can add packages but the benefit in this would be the ability to remove unwanted packages to save space and Taylor a custom firmware.  After clicking Rebuild Request it comes back to let you download a Factory or SysUpgrade.   I've seen screen shots of a textbook became available to select packages, but it doesn't come up.  I've also seen screenshots of an advanced check box in the header, it isn't there anymore.   Anyway around this?  Thanks.

aparcar, 2020-03-09:
Hi, the current Chef is no longer supported as I'm to bad at JavaScript. If anyone want to help out I'm more than happy to collaborate!

There came up a different version called [YAFS](https://github.com/mwarning/yet_another_firmware_selector) which supports about the same features, you might saw it already as it"s currently running at https://chef.libremesh.org.

To remove package add a `-` in front of the package name, like `-opkg`, but be careful that some package have dependencies so the package would be installed anyway. Like `luci` will automatically install `opkg`.

TLCary, 2020-03-10:
That will work.  Didn't realize that chef.libremesh.org was running a different version. Thanks for the heads up.  Appreciate the help.

closed 2020-03-10 by TLCary