libremesh/chef #87: TL-WDR4300-v1 builds seem to be failing

sokoow, 2020-12-03:
Hi All,

I tried to build a custom for TL-WDR4300-v1 and it's failing currently: 

```
### STDOUT

Building images for ar71xx - TP-LINK TL-WDR4300 v1
Packages: luci-app-statistics miniupnpd miniupnpc luci-mod-network luci-proto-wireguard kmod-usb-gadget-mass-storage kmod-usb-storage-extras wireguard-tools luci-mod-status kmod-wireguard luci-app-upnp atftpd luci-app-wireguard luci-app-uhttpd wireguard luci-app-wifischedule luci-mod-admin-full uhttpd luci-mod-rpc kmod-usb-storage base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-ipt-offload kmod-usb-core kmod-usb-ledtrig-usbport kmod-usb2 libc libgcc logd mtd netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic

Installing packages...
Installing libc (1.1.24-2) to root...
Installing libgcc1 (7.5.0-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libgcc1_7.5.0-2_mips_24kc.ipk.
Configuring libgcc1.
Configuring libc.
Installing kernel (4.14.195-1-342af9e4f67b3447c53216ab8e3b12a1) to root...
Configuring kernel.
Installing luci-app-statistics (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/luci-app-statistics_git-20.334.47525-e9bfb12-1_all.ipk.
Installing liblua5.1.5 (5.1.5-3) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/liblua5.1.5_5.1.5-3_mips_24kc.ipk.
Installing miniupnpd (2.2.0-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/miniupnpd_2.2.0-2_mips_24kc.ipk.
Installing kmod-nf-reject (4.14.195-1) to root...
Downloading file:packages/kmod-nf-reject_4.14.195-1_mips_24kc.ipk
Installing kmod-nf-ipt (4.14.195-1) to root...
Downloading file:packages/kmod-nf-ipt_4.14.195-1_mips_24kc.ipk
Installing kmod-ipt-core (4.14.195-1) to root...
Downloading file:packages/kmod-ipt-core_4.14.195-1_mips_24kc.ipk
Installing libip4tc2 (1.8.3-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libip4tc2_1.8.3-1_mips_24kc.ipk.
Installing libxtables12 (1.8.3-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libxtables12_1.8.3-1_mips_24kc.ipk.
Installing libip6tc2 (1.8.3-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libip6tc2_1.8.3-1_mips_24kc.ipk.
Installing iptables (1.8.3-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/iptables_1.8.3-1_mips_24kc.ipk.
Installing kmod-nf-reject6 (4.14.195-1) to root...
Downloading file:packages/kmod-nf-reject6_4.14.195-1_mips_24kc.ipk
Installing kmod-nf-ipt6 (4.14.195-1) to root...
Downloading file:packages/kmod-nf-ipt6_4.14.195-1_mips_24kc.ipk
Installing kmod-ip6tables (4.14.195-1) to root...
Downloading file:packages/kmod-ip6tables_4.14.195-1_mips_24kc.ipk
Installing ip6tables (1.8.3-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/ip6tables_1.8.3-1_mips_24kc.ipk.
Installing libpthread (1.1.24-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libpthread_1.1.24-2_mips_24kc.ipk.
Installing librt (1.1.24-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/librt_1.1.24-2_mips_24kc.ipk.
Installing libuuid1 (2.34-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libuuid1_2.34-1_mips_24kc.ipk.
Installing miniupnpc (2.1.20190408-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/miniupnpc_2.1.20190408-2_mips_24kc.ipk.
Installing libminiupnpc (2.1.20190408-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libminiupnpc_2.1.20190408-2_mips_24kc.ipk.
Installing luci-mod-network (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/luci-mod-network_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-base (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/luci-base_git-20.334.47525-e9bfb12-1_mips_24kc.ipk.
Installing libiwinfo20181126 (2019-10-16-07315b6f-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/libiwinfo20181126_2019-10-16-07315b6f-1_mips_24kc.ipk.
Installing liblua5.1.5 (5.1.5-3) to root...
Installing luci-proto-wireguard (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/ar71xx/generic/dl/luci-proto-wireguard_git-20.334.47525-e9bfb12-1_all.ipk.
Installing kmod-udptunnel6 (4.14.195-1) to root...
Downloading file:packages/kmod-udptunnel6_4.14.195-1_mips_24kc.ipk
Installing kmod-udptunnel4 (4.14.195-1) to root...
Downloading file:packages/kmod-udptunnel4_4.14.195-1_mips_24kc.ipk
Installing kmod-wireguard (4.14.195+1.0.20200611-1) to root...
[... 243 lines cut ...]
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 59260 bytes, expecting 59279 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: ppp is 98563 bytes, expecting 98560 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10326 bytes, expecting 10325 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: swconfig is 8086 bytes, expecting 8077 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7196 bytes, expecting 7198 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7580 bytes, expecting 7578 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: urandom-seed is 1539 bytes, expecting 1533 bytes
 * opkg_install_cmd: Cannot install package urandom-seed.
 * opkg_install_pkg: Package size mismatch: urngd is 5738 bytes, expecting 5736 bytes
 * opkg_install_cmd: Cannot install package urngd.
 * opkg_install_pkg: Package size mismatch: wpad-basic is 296037 bytes, expecting 296036 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [Makefile:158: package_install] Error 255
make[1]: *** [Makefile:118: _call_image] Error 2
make: *** [Makefile:208: image] Error 2
```

anyone can look at that ?

agusalex, 2021-01-25:
Same here for wdr-4300

ilario, 2021-01-26:
Ah hem, did this happen using Chef?
I think Chef is currently abandoned, is this right @aparcar? In this case, can we write this in the readme of the repository?

Currently, the only supported way to obtain the latest LibreMesh image is to compile it locally following these instructions:
https://libremesh.org/development.html

Sorry for that :/

Please try again with the supported compilation method.

aparcar, 2021-03-14:
The interface at chef.libremesh.org supports lime-packages, just enter whatever package combination you want to use there.

Please open ASU specific issues here: https://github.com/aparcar/asu/
Please open interface specific issues here: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org

Chef is no longer developed.

closed 2021-03-14 by aparcar